hawku/TabletDriver #943: TabletDriverGUI.exe Not Starting (Dissapearing)

babjakbalazs, 2020-12-28:
Can't get TabletGriverGUI.exe to start. Whenever i start it shows the process in the task manager then it says 'suspended' and closes. Installed all packages. Disabled antivirus and firewall. 

## Hardware/software information
Windows Version:  Windows 10 Pro 64-bit

Tablet Driver version:  0.2.3

Tablet model: Wacon CTL-472 Small

babjakbalazs, 2020-12-29:
I tried on laptop and it opens but its not my laptop so i dont know what to do

EternalDespair, 2021-01-10:
I have resolved this issue on my pc by going to windows update and uninstalling the security update (KB4592438)
I'm not sure if this is the reason, but every time I reinstall this update the issue returns.

babjakbalazs, 2021-01-10:
@EternalDespair Where you uninstalled it?

babjakbalazs, 2021-01-10:
nvm i find it, it might work

babjakbalazs, 2021-01-10:
its not working :/

EternalDespair, 2021-01-10:
That was the only thing I found to be causing this issue. Every time windows reinstalls that update, my tabletgui software won't load (keeps showing the icon in the taskbar for a few seconds before it disappears). Perhaps it could be a different windows update that is causing yours to not launch. I'm sorry, but I don't have any further advice.