hawku/TabletDriver #946: Can't install the devcon.exe

slenshortman, 2020-12-30:
This is what is shows

  Are you sure you want to install vmulti driver?

Press enter to install...

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>driver_vmulti\install_vmulti_64.bat

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>REM  --> Check for permissions

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>"C:\Windows\system32\cacls.exe" "C:\Windows\system32\config\system"   1>nul 2>&1

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>REM --> If error flag set, we do not have admin.

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>if '0' NEQ '0' (
echo Requesting administrative privileges...
 goto UACPrompt
)  else (goto gotAdmin  )

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>if exist "C:\Users\micah\AppData\Local\Temp\getadmin.vbs" (del "C:\Users\micah\AppData\Local\Temp\getadmin.vbs"  )

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>pushd "C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3"

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3>CD /D "C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3\driver_vmulti\"

C:\Users\micah\Downloads\TabletDriverV0.2.3\TabletDriverV0.2.3\driver_vmulti>devcon.exe /r install vmulti.inf pentablet\hid
devcon.exe failed.
Press enter...

## Hardware/software information
Windows Version: windows home edition 64 bit

Tablet Driver version: TabletDriverV0.2.3

Tablet model: CTL-4100

AWandererInInternet, 2020-12-30:
Hello, have you tried doing this?
https://i.imgur.com/8YiehgA.gif

slenshortman, 2021-01-04:
It solved, thank you.