hawku/TabletDriver #949: Laptop can't boot

SerinaSama, 2020-12-31:
So I installed this. Told me to reset. Did that. Now my laptop won't boot. GG.

## Hardware/software information
Windows Version: 10

Tablet Driver version: don't know

Tablet model: G640

negation-ren, 2021-01-05:
Would like to note that I am also experiencing a similar - if not the same - problem. Installing the drivers have caused my computer to get stuck on the loading screen when booting up. Trying to boot into safe mode left me with a pitch-black screen. I tried uninstalling the drivers using cmd but was greeted with a **0xE0000302** error, so I deleted the files altogether. Didn't bring any results, still stuck in a loop when booting at the moment, will retry in the morning. I'm not trying to be a dickhead by barging into OP's problem, but I just thought this experience might be helpful in the process of resolving this problem.

OP, have you tried forcing the laptop to auto repair (turning it on and off 2+ times until it starts auto repairing) => Advanced Options => Troubleshoot => Advanced Options => Startup Settings, and then choosing to boot into safe mode to remove the drivers? Alternatively, choosing command prompt instead and trying to uninstall the drivers from the cmd?

SerinaSama, 2021-01-07:
Hello! I've actually fixed the problem. My windows was corrupted, so I had to use a boot up usb to install a new windows