hawku/TabletDriver #961: VCRUNTIME140.dll and MSVCP140.dll is missing

Guill4um304, 2021-01-12:
When I start TabletDriverGUI.exe, it show me VCRUNTIME140.dll is missing and when I close it, it show me MSVCP140.dll is missing. Someone can help me ?

## Hardware/software information
Windows Version: 

Tablet Driver version: 
V0.2.3

Tablet model: 
Wacom CTL-4100

DemonDollOW, 2021-01-23:
same problem here, i tried reinstalling wacom drivers and retrying all the process for the driver but nothing changed

Guill4um304, 2021-01-24:
I have found the solution. Install this. It's visual studio redistribuable x64 and x86.
[vc_redist.x86.zip](https://github.com/hawku/TabletDriver/files/5861806/vc_redist.x86.zip)

[vc_redist.x64.zip](https://github.com/hawku/TabletDriver/files/5861805/vc_redist.x64.zip)

Guill4um304, 2021-01-24:
I have found the solution. Install this. It's Visual Studio Redistribuable x64 an x86. If it's already install click on ''Repair'' button after executing the .exe file.

[vc_redist.x64.zip](https://github.com/hawku/TabletDriver/files/5861911/vc_redist.x64.zip)
[vc_redist.x86.zip](https://github.com/hawku/TabletDriver/files/5861912/vc_redist.x86.zip)

MajorDK, 2021-02-04:
Didn't work for me. Even after libraries download anr reinstallation of a driver. This error message is still poppin up.

DIGHEAD, 2021-05-15:
I had the same issue, but I downloaded the 2 files from Guill and it appears to work. Before that I did an entire P.C reset which may or may not have helped.

closed 2023-02-21 by Guill4um304