hawku/TabletDriver #977: Cursor dont move.

kabuzito, 2021-02-01:
Cursor dont move, appear ''Connected'' but the cursor dont move.

## Hardware/software information
Windows Version: 10

Tablet Driver version: v0.2.3

Tablet model: Huion H430p

Teddie910, 2021-02-02:
Agreed. Console says my tablet is connected by when I hover my pen over my tablet it doesn't move. Never had this issue before, I have used this driver in the past and am getting back into osu after a while.

SetSpawn, 2021-02-07:
Same i also got the same problem i can't use it when it's on absolute but when i switch it to debug it works but the area i set doesn't register.

Tablet: VEIKK S640

alixtrixia, 2021-02-11:
Have a similar issue where my tablet and pen is connected (shows my pen movements in the console) but the cursor does not move at all.

Tablet: Wacom CTL-4100