hawku/TabletDriver #979: Can't set area

davidkiko, 2021-02-01:
I am unable to set a custom tablet area. No matter what I do, it just stays full area. I've tried reinstalling the drivers, still no luck. I don't have any other tablet drivers installed.

## Hardware/software information
Windows Version: 10.0.18363 Build 18363

Tablet Driver version:  0.2.3

Tablet model: CTL-472

davidkiko, 2021-02-01:
https://imgur.com/a/SazVqVz

kerabite, 2021-02-02:
Please elaborate. Are you unable to change the values of the tablet area or does the tablet area not seem to change?

If it is the latter, ensure that you are clicking "Save settings"; if you don't click that, your settings will not seem to update.

davidkiko, 2021-02-02:
The area didn't change no matter what I set it to. I did click save settings after every try.

Fixed the issue by downloading older drivers (v0.2), for some reason those worked fine. No clue what caused the current version to not work.

XCPA26X, 2021-02-03:
I have the same problem, I have to use full area and I did everything necessary to try fix it. although i think it has something to do with another driver interfering.

XCPA26X, 2021-02-03:
lol just fixed it. i just restarted the console

closed 2021-02-07 by davidkiko