hawku/TabletDriver #982: Can't set area, stays on full yet the speed the pen moves at increases.

Ctl472wtfitdotho, 2021-02-03:
So I just bought my ctl-472 yesterday. I could change area etc. Then an hour or so later it reset to full area and whatever I do it doesn't change. I went from the hawku drivers to the opentablet ones and back to wacom. Idk what to do.
Some previous post said that going back to v0,2 resolved his issue but it didn't help with mine. 

## Hardware/software information
Windows Version:  Windows 10

Tablet Driver version:  v0,2

Tablet model: ctl-472

Redtern573, 2021-02-06:
Same. I even go back to v1.5 bit still

Ctl472wtfitdotho, 2021-02-07:
> Same. I even go back to v1.5 bit still

Yeah, I fixed it by switching drivers. I just installed the devocube version of hawku and it worked.
Goodluck on fixing it for yourself tho.

closed 2021-02-07 by Ctl472wtfitdotho