hawku/TabletDriver #990: [XP PEN - G640S] Tablet init failed

Thomaska111, 2021-02-12:
Anyone know how to fix it?
[image]

## Hardware/software information
Windows Version: Windows 10 Pro 1909

Tablet Driver version: 0.2.3

Tablet model:XP PEN G640S

koutyy, 2021-02-12:
Find XP-Pen G640s in TabletDriverV0.2.3\config\tablet.cfg
then change InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 to InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 and restart driver
Maybe this will help

Thomaska111, 2021-02-13:
> Find XP-Pen G640s in TabletDriverV0.2.3\config\tablet.cfg
> then change InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 to InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 save changes and restart driver
> Maybe this will help you

give another error
[image]

koutyy, 2021-02-13:
> 
> 
> > Find XP-Pen G640s in TabletDriverV0.2.3\config\tablet.cfg
> > then change InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 to InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 save changes and restart driver
> > Maybe this will help you
> 
> give another error
> [image]

Try change ReportLength to 12 and replace InitReport with InitReport 0x02 0xB0 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00