hawku/TabletDriver #994: Linux support request

Maritsu, 2021-02-17:
So I've tried the official XP-Pen drivers but they're unusable and I remembered these drivers exist. I'm not a Windows user though and I think these drivers are the only ones that would support my tablet but I can see there's no Linux version. So I'd request Linux support for these drivers.

## Hardware/software information
~~Windows Version:~~
**Linux Version:** Ubuntu 20.04 LTS

**Tablet Driver version:** None, as there is no Linux support yet

**Tablet model:** XP-Pen Star G430S

tilda, 2021-02-19:
OpenTabletDriver seems to fully support your tablet and is cross-platform. https://github.com/InfinityGhost/OpenTabletDriver

Maritsu, 2021-02-19:
Alright, thanks for help

closed 2021-02-22 by Maritsu