hawku/TabletDriver #995: H640P | Unknown Huion Tablet!

Charchah, 2021-02-20:
I looked around on here and saw people with similar issues from like a year ago. Just wondering if there was a solution I missed or if the solution has been solved already and I'm just inept? Basically the same thing is happening, I have the tablet connected, uninstalled and reinstalled drivers, removed OEM Drivers. Restarted my computer in-between uninstalls and installations. Am I missing something here? I tried following tips from reddit and other posts here and I'm hella lost. It detects the tablet being there but not as it should be. Tablet doesn't function and here we are.

[startuplog.txt](https://github.com/hawku/TabletDriver/files/6014144/startuplog.txt)

## Hardware/software information
Windows Version: Windows 10 Enterprise 19H2

Tablet Driver version: TabletDriverV0.2.3

Tablet model: Huion H640P

SesKaza, 2021-03-14:
I just found a fix for this today, first you have to open your tablet.cfg file and delete "CheckString 201 "HUION_T173"". This work for me, i hope it will work for you

Gittywittyy, 2021-06-21:
it did not work =(

Gittywittyy, 2021-06-21:
> I just found a fix for this today, first you have to open your tablet.cfg file and search for Huion H640P and you should delete "CheckString 201 "HUION_T173"". This work for me, i hope it will work for you

this wont work but this will https://www.reddit.com/r/osugame/comments/kj25u1/for_whoever_is_having_problems_with_the_hawku/