hawku/TabletDriver #999: Smoothing filter adds incredible lag

messygamermode, 2021-02-26:
If I change any of the values at all, I can see noticeable lag. It wasn't like this before, it just randomly started doing it, and with it it's unplayable.

## Hardware/software information
Windows Version: 

Tablet Driver version: 0.3.1

Tablet model: Gaomon S620

pinballmanfan, 2021-03-14:
This just started happening for me too. What tablet do you have and what version of Windows are you on?

Blast1113, 2021-03-14:
I've had this for a few months now.
I've tried every tablet I own (CTH480, CTH670, G640, H420) and they all have this problem. From some rudimentary testing it appears to be stuck at 60Hz, no matter what 'Rate' setting I use.

### Settings
(With extra info)
[image]
[image]