hawku/TabletDriver #1000: XP-Pen Star G640 not being recognized

drecho408, 2021-02-28:
Pen is "recognized" by software but not being recognized by software... any ideas?

## Hardware/software information
Windows Version: Windows 10
Tablet Driver version: 0.2.3

Tablet model: Star G640
[image]

GamerTater, 2021-03-07:
I'm dealing with the same problem. Found anything yet?

Ponymous, 2021-03-08:
@GamerTater @drecho408 both of you just got scammed. XP-PEN is selling a revision of their tablet that isn't compatible with hawku drivers and has a shit ton of smoothing, see this https://www.reddit.com/r/osugame/comments/gwnrpv/important_xp_pen_released_a_new_g640/

diezz1kk, 2021-05-09:
Do you guys know why the xp pen g640 isnt being detected by my pc like i have nothing i downloaded the driver and yet nothing  no detection please message me on insta its.justangel