hawku/TabletDriver #1028: Driver just stopped working without any reason

Ankokukun, 2021-03-17:
## Hardware/software information
Windows Version: 10 

Tablet Driver version: 0.2.3

Tablet model: Gaomon S620

The Driver worked completely fine at first but after two days just gave up without any error message. When starting it just says: driver stopped. Restarting! Check console!!!

Heres: all the code appearing in the cosole:
[image]
[image]
[image]

HolyNexus, 2021-03-18:
same but my tablet is from wacom :/

FreeHK-Lunity, 2021-03-23:
#1024