hawku/TabletDriver #1033: The driver only works for me if the Wacom drivers are uninstalled

FeatsM, 2021-03-20:
I tried using the "DisableWacomDrivers" and it did close everything of my Wacom tablet, but in the console of the TabletDriver it says:
[image]
this happens when the Wacom driver is installed (even when I used the "DisableWacomDrivers"). After uninstalling the Wacom drivers the TabletDriver does work by putting this in the config/tablet.cfg file:
[image]
But I need to restart my PC every time I install/uninstall the Wacom driver and that's pretty annoying if I want to draw and then switch to osu. I saw other people having the same problem but for them the "DisableWacomDrivers" tool does work. I think it's only happening with the CTL-6100.
If anyone knows a (temporary) fix to this I would appreciate it!

## Hardware/software information
Windows Version: 
Windows 10 
Tablet Driver version: 
v0.2.3
Tablet model: 
Wacom Intuos CTL-6100

adryzz, 2021-03-21:
use OpenTabletDriver and switch between configuration files

FreeHK-Lunity, 2021-03-23:
#1024

FeatsM, 2021-03-23:
OpenTabletDriver works for me thank you both :)

closed 2021-03-23 by FeatsM