hawku/TabletDriver #1040: H640p not working

BoricSpider, 2021-03-24:
## Hardware/software information
Windows Version: windows 10 

Tablet Driver version: v0.2.3

Tablet model: H640p

Recently the hawku driver stopped detecing my tablet. When i run debugging it tells me that it reads my pen and all but it won't move. The console is also saying Unkown Huion Tablet and its getting on my nerves of not being able to fix this. Please help me

[image]

[image]

adryzz, 2021-03-28:
#1024

Gittywittyy, 2021-06-21:
Yo here is a fix https://www.reddit.com/r/osugame/comments/kj25u1/for_whoever_is_having_problems_with_the_hawku/