hawku/TabletDriver #1045: No apply button - does this mean the tablet has not connected?

hp-well, 2021-04-01:
Hello,
I have recently bought a tablet and I am trying to set it up with a good driver for osu! . When I open the driver, I can change the area and buttons, but it does not change anything for the tablet - e.g. it still uses full area after me disabling that. Furthermore, there is no apply button as I have seen when other people use it. Many thanks in advance.

## Hardware/software information
Windows Version: 
Windows 10
Tablet Driver version: 
0.2.3 
Tablet model: Huion 420

SnowLire, 2021-04-08:
#1024

FreeHK-Lunity, 2021-04-27:
Your best bet is to stick with Huion's driver since the only other community driver OTD doesn't support the Huion 420.