hawku/TabletDriver #1054: XP-PEN Deco 01 v2 not detected

RandomPotato0, 2021-04-08:
Hey, I just bought this tablet and tried to install the drivers but I'm constantly getting this error

[2021-04-08 17:06:08.636] >> CheckTablet
[2021-04-08 17:06:08.636] [ERROR] Tablet not found!
[2021-04-08 17:06:08.636] [ERROR] Check the list of supported tablets from the GitHub page.
[2021-04-08 17:06:08.636] [ERROR] http://github.com/hawku/TabletDriver
Driver stopped. Restarting!

This model is supposed to be working but I keep getting this error, I just want to change the tablet area, cheers

## Hardware/software information
Windows Version: 10

Tablet Driver version: V0.2.3

Tablet model: XP-PEN Deco 01 v2

SnowLire, 2021-04-08:
#1024

RandomPotato0, 2021-04-08:
> #1024

oh well, thanks for letting me know