hawku/TabletDriver #1060: Driver uses two monitors when selecting one display

dbot101, 2021-04-17:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: wacom CTL-490

I have reinstalled TD, restarted my computer many times, etc etc.

Set area registers that I have two different monitors but no matter what I do the driver NEVER wants to save my settings and trys to use full area on two monitors. I set the area to my main display and W/H to 45mm, click save a million times, restart the driver, click save again and nothing changes. 
[image]
This is what it looks like, it looks normal like how I prefer it to be but in all reality it refuses to use those settings and never changes out of full area using dual monitors

I hope someone can help with my issue soon

SnowLire, 2021-04-17:
#1024

dbot101, 2021-04-17:
For anyone else experiencing this:

Install the new version of the driver #1024

closed 2021-04-17 by dbot101

adryzz, 2021-05-18:
> For anyone else experiencing this:
> 
> Install the new version of the driver #1024

it's not a new version, it's a new driver made by other people