hawku/TabletDriver #1071: Relative mode It does not change the functioning of the tablet

OkamiDarakino, 2021-04-26:
## Hardware/software information
Windows Version: 10 single language (Version 2004)

Tablet Driver version: 0.2.3

Tablet model: huion New 1060 plus

The program detects my tablet, adjust the area correctly, but when selecting the "relative" mode there is no change at all, it works exactly like the absolute mode or that the windows ink, the program if it detects my tablet without problems, but does not work like a mouse at any time.

NotLoki11, 2021-04-26:
This could be a fix for me it was they way i held my pen when it was at a big angle i would hold it more straght up or u need to save the settings

SnowLire, 2021-04-27:
most likely scenario is that some windows update broke the driver. you can try [OpenTabletDriver](https://github.com/OpenTabletDriver/OpenTabletDriver/releases) if that doesn't work then just use Huion's driver