hawku/TabletDriver #1076: Problems between drivers / installations

ToxicPeople, 2021-04-30:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: v14.8.165.1387

Tablet model: Huion Inspiroy H950P

Today I had a problem, the inspiroy h950p graphics tablet has worked perfect for me to play osu until today, the issue is to update the driver version since an update came out. The issue is that I could not continue playing, reinstalled the version I was using but it didn't work either, then I tried to reinstall the tablet driver for osu and it didn't work either, I have no idea what to do. Until today it worked correctly and without any problem, I know that it is not within the supported versions but it would be very helpful some support because I am really sad if I can no longer continue playing.

ToxicPeople, 2021-04-30:
[image]

ToxicPeople, 2021-04-30:
[image]

Ponymous, 2021-05-01:
this repo is dead, you wont find any people willing to fix your problem