hawku/TabletDriver #1087: [PSA] Stop using this driver

SnowLire, 2021-05-15:
Hawku TabletDriver has been abandoned for 3 years, and counting, and should not be used. Windows Update will break functions in the driver and it will stop working. And don't even think about support for newer tablets.

Please switch over to [OpenTabletDriver](https://github.com/OpenTabletDriver/OpenTabletDriver) and check if [your tablet is compatible](https://github.com/OpenTabletDriver/OpenTabletDriver/projects/4). If your tablet is not supported by OTD, use the OEM driver instead until you have a better tablet.

KanChi576, 2021-05-20:
and what now ? In that case i can't use the driver anymore ? i install the driver again and then it says " Error! Can't start the driver!
bin\TabletDriverService.,exe not found "

SnowLire, 2021-05-21:
> and what now ? In that case i can't use the driver anymore ? i install the driver again and then it says " Error! Can't start the driver!
> bin\TabletDriverService.,exe not found "

Use OTD, please read the entire post, thanks!

ryunosuke1121, 2021-06-14:
To everyone having issues with the cursor only covering the screen in-game partially or half the screen when you are playing Osu!: 

Set your Display Area to all of your monitors/displays and the issue will be fixed in-game.

@SnowLire this fix also works for [OpenTabletDriver](https://github.com/OpenTabletDriver/OpenTabletDriver) it was having the same issue (for me).

SnowLire, 2021-06-15:
OTD's version of VMulti is getting fixed, so this won't be a lasting problem, only for Hawku/Devocub

jxu, 2021-12-20:
Still works fine for windows 8.1 and my ancient huion 608

VB304, 2022-03-30:
Reinstalled windows 10 and did the same configuration and still works.

I don't know which one that did it but I did:

Fully enabled the .Net framework 3.5 and 4.8 in windows feature.
Installed veikk drivers but not in Startup

[image]

jamesbt365, 2022-06-12:
b

Phantom-bits, 2022-07-10:
c

VB304, 2022-07-11:
d

jamesbt365, 2022-07-13:
e

Artrix9095, 2022-09-16:
f

dotSepia, 2025-09-18:
g