hawku/TabletDriver #1093: Hawku tablet driver area not working correctly

Rvvni, 2021-05-22:
I ordered a new tablet because I thought my Viekk was just broken, but the same exact thing is happening on my new Wacom.
When I set an area the tablet is really inaccurate and when I take it out of the area it get's stuck to the side and I can only move it up and down when it is stuck to the side. I have been struggling with this for the past 5 months. I just want someone to help me. Please...

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: Viekk S640 & Wacom Ctl-472

SnowLire, 2021-05-22:
#1087

totalbummer, 2021-06-10:
Copypasting my own response from a different thread. Hope this helps! I have dual displays so idk if this will be the same process for you/what your setup is.

"Here for the same reason the 4 posters above are here. Windows update that I didn't ask for ruined the fun. Hoping for a workaround.

Second update: When I finished playing osu, I re-enabled my second display by re-selecting extend these displays and the tablet still covers the entire main display area. My guess is until I restart.

Update: I got the tablet functioning on my Main Display as intended but had to set display properties to show only main display and not extend/duplicate, effectively disabling my second display while using the driver.

After I changed that setting, I unchecked "auto" on the Desktop Size field and manually entered my display resolution, hit apply. Nothing changed. I re-checked "auto," hit apply again, and then it worked. I hope this helps some of you, Win10/CTL-472 here."

SnowLire, 2021-06-12:
use OpenTabletDriver