hawku/TabletDriver #1109: Drivers only working on half of monitor after Windows Update

Brodod, 2021-06-11:
I updated to the most recent Windows 10 version last night and when I opened the tablet drivers this morning it would only let me move my cursor using my tablet on half of my monitor. I know that I am not the only one having this issue as there were multiple people that posted on r/Osugame after the Windows update having the same problem too.

I read some other posts on here about a similar problem but after reading I realized that these people could still properly use their tablet on their desktop, just not in Osu!. This problem occurs on my desktop, not just the game.

## Hardware/software information
Windows Version: 10 Home 64 bit

Tablet Driver version: v0.2.3

Tablet model: XP-Pen G640

XZiuS, 2021-06-11:
me too model wacom CTL 472

VeakeR, 2021-06-11:
Same here, Wacom CTL 472

P3di4s, 2021-06-11:
I have the same problem, but with this settings i can still play, dont know why, but this works for me. Display 2 is my main Display and the one I play osu! on[image]
[image]

XZiuS, 2021-06-11:
[image]
i tried this, and it works for me oon my monitor 1

XZiuS, 2021-06-11:
[image]
i tried this and it works for my 1st monitor
the sensitivity is weird i still cant play

BadatHoppin, 2021-06-11:
Im currently having the same problem, Huion H430P

Troved, 2021-06-11:
this also happened to me after the windows update and my tablet is the CTL-472. I still have no idea how to fix it hopefully its's fixed soon.

UltayJr, 2021-06-11:
Exact same, for 2 days now.
Windows version: 10 pro, 2004 version and 19041.1052 build
Tablet driver version: 0.2.3
Tablet model: CTL-480
The latest windows update is the problem, on a 3-monitor setup, the same problem, my area is a square at the middle of my main screen and can't be modified
[image]

sambouiss, 2021-06-11:
seeting 

> I have the same problem, but with this settings i can still play, dont know why, but this works for me. Display 2 is my main Display and the one I play osu! on[image]
> [image]

This worked for me god bless

neckztik, 2021-06-11:
There is a fix for this, all you do is you need to keep your area the same and make the screen map to both your monitors.
[image]
Hope this gets fixed, but it honestly probably won't.

TheMinecar, 2021-06-11:
Same thing is happening to me except I'm on a VEIKK S640.

Instellate, 2021-06-11:
> I updated to the most recent Windows 10 version last night and when I opened the tablet drivers this morning it would only let me move my cursor using my tablet on half of my monitor. I know that I [...]

Hey everyone that reads this. I will just tell this. The issue is made by the new windows security update. I uninstalled the update and the drivers worked fine. I do not recommend to uninstall it if you dont know what you are doing. Until that, just wait until its fixed or uninstall the security update if you know what you are doing!

SnowLire, 2021-06-11:
#1087

SnowLire, 2021-06-11:
> Hey everyone that reads this. I will just tell this. The issue is made by the new windows security update. I uninstalled the update and the drivers worked fine. I do not recommend to uninstall it if [...]

this will never get fixed btw. hawku has abandoned this project years ago

AyanoNK, 2021-06-11:
> t y

I will try to peek at the files and see if I can fix it. I'll make a branch if I can fix it.

SnowLire, 2021-06-12:
> I will try to peek at the files and see if I can fix it. I'll make a branch if I can fix it.

why bother, just use OpenTabletDriver

AyanoNK, 2021-06-12:
> why bother, just use OpenTabletDriver
:(

Troved, 2021-06-12:
how do you fix this please I want to play

SnowLire, 2021-06-12:
> how do you fix this please I want to play 

you don't. use OpenTabletDriver

danielkemp017, 2021-06-12:
Quick Tip guys a current workaround i found was to set ur area as per normal but set the 'set area' dropdown to full desktop 
[image]
after that go into the Misc tab then set the desktop to whatever size is ur native in my case 1920*1080 
[image]
Hope this helps until a more permanent fix is done

lukemagdalin, 2021-06-12:
> Quick Tip guys a current workaround i found was to set ur area as per normal but set the 'set area' dropdown to full desktop
> [image]
> after that go into the Misc tab then set the desktop to whatever size is ur native in my case 1920*1080
> [image]
> Hope this helps until a more permanent fix is done

Thanks, This works for the time being.

Troved, 2021-06-13:
> > how do you fix this please I want to play
> 
> you don't. use OpenTabletDriver

I tried using it but it keeps telling me to download .NET and I already did but it still tells me to download it. 
[image]

SnowLire, 2021-06-13:
> I tried using it but it keeps telling me to download .NET and I already did but it still tells me to download it
> [image]
> 
> I already pressed yes and downloaded it
> and nothing happens.

was the file named windowsdesktop-5.0.7-win-x64? if not then you probably downloaded the wrong one. You have to download .NET 5 Desktop Runtime x64

Instellate, 2021-06-13:
> this will never get fixed btw. hawku has abandoned this project years ago

Nah. Microsoft will fix it cause its a problem with multiple drivers. So we just need to wait for them to fix it next week

XZiuS, 2021-06-13:
> > > how do you fix this please I want to play
> > 
> > 
> > you don't. use OpenTabletDriver
> 
> I tried using it but it keeps telling me to download .NET and I already did but it still tells me to download it
> [image]
> 
> I already pressed yes and downloaded it
> and nothing happens.
 same here

XZiuS, 2021-06-13:
> > > how do you fix this please I want to play
> > 
> > 
> > you don't. use OpenTabletDriver
> 
> I tried using it but it keeps telling me to download .NET and I already did but it still tells me to download it
> [image]
> 
same here

SnowLire, 2021-06-13:
[.NET 5 Desktop Runtime](https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.7-windows-x64-installer)

Troved, 2021-06-14:
> [.NET 5 Desktop Runtime](https://dotnet.microsoft.com/download/dotnet/thank-you/runtime-desktop-5.0.7-windows-x64-installer)

ty but after this how do u fit it in one monitor only? i have two of em and it only works on my second one rn
[image]

In my old devocub one it has an option
[image]