hawku/TabletDriver #1116: Tablet only using certain section of my screen.

lawrencessj24, 2021-06-12:
## Hardware/software information
Windows Version: 
Windows 10
Tablet Driver version: 
0.2.3
Tablet model: veikk s640

I have 3 monitors which I believe is part of the issue. When I switch the set area to a certain monitor its splits my main screen into three. When I set the area to my left monitor, it only lets me use the left third of my middle monitor and etc. How can I fix this problem?

Mrcubix, 2021-06-13:
vmulti broke in the latest windows update, either downgrade or switch to OTD #1087

Z4ckFairX, 2021-06-13:
The solution i found is to make the area the total of you screens. I have a 1920 and 1440 screens so i set it to 3360 manually. It seems to be working as it only uses the main screen.
[https://i.imgur.com/wrpdsND.png](url)