hawku/TabletDriver #1117: hello I need help with the deinstallation

lxstkrieger, 2021-06-13:
when i click on remove_vmulti_driver to uninstall hawku it doesn't work. I then tried to put the files in the trash, but unfortunately that didn't work either
(sry for my english )

## Hardware/software information
Windows Version: 
Windows 10 Home
Tablet Driver version: 
0.2.3
Tablet model: 
wacoom-CTL-4100

lxstkrieger, 2021-06-13:
I have another 2 problem. my cursor can't go further than the middle of the screen if that's fixed i don't need to uninstall the hawku driver either

Mrcubix, 2021-06-13:
windows update broke hawku #1087

lxstkrieger, 2021-06-14:
thank you for telling me i'm switched to the open tablet driver now, it feels a bit strange but i think that's normal when you change drivers 

(google translator) :)

lxstkrieger, 2021-06-14:
but somehow i can't cope with the open tablet driver that is a different feeling than with the hawku driver it is normal

closed 2023-10-27 by lxstkrieger