hawku/TabletDriver #1119: Drivers in absolute mode don't allow me to use all of my main display

kairosrulz, 2021-06-13:
I have a G640 tablet and whenever I put the drivers into absolute mode I can't move my cursers past a small area on my main monitor. I tried to reinstall but it still hasn't fixed the issue. any advice would be helpful.

## Hardware/software information
Windows Version: 10

Tablet Driver version: V0.2.3

Tablet model: G640
[image]

Suinollat, 2021-06-13:
I have the same issue with my Veikk S640, It will not map correctly. This happened directly after the new Windows 10 update. I had played osu! before putting my computer to sleep, and windows updated when I woke up (Thanks Microsoft...) and it will not work now. I reinstalled drivers, restarted, and changed settings and it won't work.

kairosrulz, 2021-06-13:
https://github.com/OpenTabletDriver/OpenTabletDriver differnt dirvers work better
@Suinollat

ryunosuke1121, 2021-06-16:
Hi, uncheck the force aspect ratio in your tablet settings and set the area to both displays to fix the issue.

Windows changed the way they interact with tablets and this is the only workaround we have so far.