hawku/TabletDriver #1121: Cursor confined to left side of screen

br1cck, 2021-06-14:
I haven't changed any settings and everything was working fine 2 or 3 days ago. For some reason when I turned my computer on today and tried to play osu I couldn't move my cursor past the middle of my screen, it's stuck on the left side. Tried restarting drivers, restarting computer, unplugging/replugging my tablet, and completely reinstalling drivers. Worked fine in lazer (I think because of the built in drivers). 
[image]

## Hardware/software information
Windows Version: 10 (10.0.19041 build 19041)

Tablet Driver version: 0.2.3

Tablet model: Wacom CTL-4100

SnowLire, 2021-06-14:
#1087

ryunosuke1121, 2021-06-14:
To everyone having this issue:

Set the Display Area to all of your monitors and the issue will be fixed in Osu!

This is also an issue with OpenTabletDrivers (Updated Hawku drivers) and the fix is the same.

Mrcubix, 2021-06-15:
> To everyone having this issue:
> 
> Set the Display Area to all of your monitors and the issue will be fixed in Osu!
> 
> This is also an issue with OpenTabletDrivers (Updated Hawku drivers) and the fix is the same.

OpenTabletDriver is a whole different driver, not the shit show that hawku is right now with everyone trying to tape patch issue every windows update. maintained, and i'll be there ass well as other individuals to help you make your tablet work, provided it's not one with known issue or too old to be supported, see https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/TABLETS.md for compatibility