hawku/TabletDriver #1122: tablet error

idkwhy1, 2021-06-14:
tablet model xp-pen g640
works only left side of the tablet, when i put pen on the right side on the monitor it is only at the middle of the display

## Hardware/software information
Windows Version: 

Tablet Driver version: 
[image]

Tablet model:

ryunosuke1121, 2021-06-14:
To everyone having this issue: 

Set the Display Area to all of your monitors and the issue will be fixed in Osu!

This is also an issue with OpenTabletDrivers (Updated Hawku drivers) and the fix is the same.

ThatGiantSeth, 2021-06-14:
> To everyone having this issue:
> 
> Set the Display Area to all of your monitors and the issue will be fixed in Osu!
> 
> This is also an issue with OpenTabletDrivers (Updated Hawku drivers) and the fix is the same.

Having exact same issue, also after a Windows update. This fix kind of worked for me, but moving up and down is extremely sensitive, and a good quarter of the tablet is offscreen. I have the Huion 420, which doesn't work properly with OpenTabletDrivers on Windows

Mrcubix, 2021-06-15:
> To everyone having this issue:
> 
> Set the Display Area to all of your monitors and the issue will be fixed in Osu!
> 
> This is also an issue with OpenTabletDrivers (Updated Hawku drivers) and the fix is the same.

OpenTabletDriver is a whole different driver, not the shit show that hawku is right now with everyone trying to tape patch issue every windows update. maintained, and i'll be there ass well as other individuals to help you make your tablet work, provided it's not one with known issue or too old to be supported, see https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/TABLETS.md for compatibility

ryunosuke1121, 2021-06-16:
> > To everyone having this issue:
> > Set the Display Area to all of your monitors and the issue will be fixed in Osu!
> > This is also an issue with OpenTabletDrivers (Updated Hawku drivers) and the fix is the same.
> 
> ~Having exact same issue, also after a Windows update. This fix kind of worked for me, but moving up and down is extremely sensitive, and a good quarter of the tablet is offscreen. I have the Huion 420, which doesn't work properly with OpenTabletDrivers on Windows~
> 
> Edit: I messed around with the active tablet area for a while and got it back to almost normal, sorry for the extra comment. For anyone else with the same tablet, you have to set the driver to both monitors like above, but set the tablet active area to just 1 monitor.

What I did was unchecked the "Force Aspect Ratio" and used a custom Width and Height to simulate a 16:9 screen. You can use a calculator to get the desired results. I own a Wacom 480 and use a width of 25 and height of 14 and it works wonders for me.

ryunosuke1121, 2021-06-16:
> OpenTabletDriver is a whole different driver, not the shit show that hawku is right now with everyone trying to tape patch issue every windows update. maintained, and i'll be there ass well as other [...]

I agree that the Hawku driver's code is kind of all over the place, but it is still incredibly popular. Also, the issue was happening in both drivers and I personally use the Open Tablet Driver to play and I managed to fix it that way (As I shared above).

SnowLire, 2021-06-16:
> I agree that the Hawku driver's code is kind of all over the place, but it is still incredibly popular. Also, the issue was happening in both drivers and I personally use the Open Tablet Driver to [...]

This is purely a VMulti problem, which Hawku depends on for its output method. If you use the base OpenTabletDriver output mode, no such thing will happen. What's most likely is that you used OTD's VMultiMode, which is only really there to make it work with apps running as admin, OTD's Windows Ink, which is used more for drawing, both uses VMulti, which is broken.