hawku/TabletDriver #1123: Osu tablet locking on to only half the screen

QuazzyG, 2021-06-14:
I recently just did a update on windows and reinstalled the tablet driver and my tablet is stuck on like half of the screen while I am on osu i think this is an issue with my two monitors is there any way to fix this

https://user-images.githubusercontent.com/85898692/121937206-56a32480-cd42-11eb-9ad4-4a22aabde48c.mp4

## Hardware/software information
Windows Version: 
10

Tablet Driver version: 
TabletDriverV0.2.3 
Tablet model: Wacom ctl 472

MysticalWizard, 2021-06-15:
I have the same issue as well... The only fix I could find for this issue was to change display settings to single screen (windows + P and select PC screen only) then relaunch the TabletDriver, make sure only one screen is shown on the driver, press save settings, and restart driver. After that change display settings to multiscreen (windows + P and select extend). Once you've changed your display settings back to multiscreen, don't touch anything and just minimize TabletDriver to tray.
The only downside of this is that you have to repeat this every time you restart your PC which is annoying but it seems to be the only working solution for now.
[image]

SnowLire, 2021-06-15:
#1087

ryunosuke1121, 2021-06-16:
Hi, uncheck the force aspect ratio in your tablet settings and set the area to both displays to fix the issue. 

Windows changed the way they interact with tablets and this is the only workaround we have so far.

GamezombieCZ, 2021-06-19:
Unchecking the force aspect ratio didn't worked for me, but setting the screen map area on full desktop did the trick for me.

xscque, 2021-09-18:
> Hi, uncheck the force aspect ratio in your tablet settings and set the area to both displays to fix the issue.
> 
> Windows changed the way they interact with tablets and this is the only workaround we have so far.

TYSM

SnowLire, 2021-09-18:
Just screenshot your settings and switch to OpenTabletDriver. Workarounds like this are just gonna become more and more prevalent in order to run this 3-4, and counting, year-old piece of software.

longxia01, 2022-04-08:
I set my 'set area' to my 'main' monitor but that didn't work. setting it to 'full desktop', did however. so yea, did take some playing with.
XPPen ( https://www.xp-pen.com/forum-2450.html ) drawing tablets are very cheap. They perform just as well as a Wacom, but much cheaper. the XPPen Star G640 is a beginner level tablet from them that is considered really good for osu too. I'd recommend that one.

SnowLire, 2022-04-08:
I'd much rather trust an [independant 3rd party's buying guide](https://docs.google.com/spreadsheets/d/1DYVfiSpQqdpa4sWWYUALPmliOIuGyKog7B7LJJdmlhE) than one made by a tablet manufacturer, which is a clear conflict of interest due to the fact that they want to make their tablets seem better than they actually are.