hawku/TabletDriver #1130: What should I do?

blazexux, 2021-06-18:
[2021-06-18 17:30:55.633] [ERROR] 1) Install VMulti driver
[2021-06-18 17:30:55.633] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-18 17:30:55.633] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-18 17:31:01.200] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-18 17:31:01.201] [ERROR] Can't open VMulti device!r
Driver stopped. Restarting!
It comes out like this, what should I do?

SnowLire, 2021-06-18:
#1087

Kiried-Menacing, 2021-06-21:
Have you ran the install_vmulti_driver.bat file or
install_vmulti_driver32.bat
The "install_vmulti_driver.bat" is for 64 bit systems
The "install_vmulti_driver32.bat" is for 32 bit systems
If you have installed the wrong vmulti driver there are remove files so you can redo it :)