hawku/TabletDriver #1131: XP PEN STAR G430s not detected by driver

bbhard, 2021-06-18:
I tried installing this driver today, but it won't get detected by the driver. Device manager detects it. 
[image]

I tried running the guy as administrator, but it wouldn't fix the issue. What else can I do to fix this?

## Hardware/software information
Windows Version: 21H1

Tablet Driver version: 0.2.3

Tablet model: xp pen g430s

SnowLire, 2021-06-18:
#1087

Kiried-Menacing, 2021-06-21:
Have you ran the install_vmulti_driver.bat file or
install_vmulti_driver32.bat
The "install_vmulti_driver.bat" is for 64 bit systems
The  "install_vmulti_driver32.bat" is for 32 bit systems

SnowLire, 2021-06-22:
no, just, uninstall hawku. the vmulti included is broken on 21H1, use OpenTabletDriver