hawku/TabletDriver #1132: Driver's picking only half of the screen after 21H1 update

MidKnightXI, 2021-06-19:
## Hardware/software information
Windows Version: 21H1 build 19043, 1055

Tablet Driver version: 0.2.3

Tablet model: Wacom CTH 480

SnowLire, 2021-06-20:
#1087

MidKnightXI, 2021-06-20:
I'm aware that hawku isn't updated anymore and even if opentablet is newer, TabletDriver is still more convenient to use, more user friendly and better (in my opinion) so why not giving just an update to fix this issue ?

SnowLire, 2021-06-20:
Because hawku said he's stopping development and maintenance 3+ years ago. Without him merging the PRs it'll never get fixed in the main branch, and the situation won't get better as more stuffs, that TD depends on, in Windows get removed you'll need more workarounds to fix the ever growing issues list.

closed 2022-04-17 by MidKnightXI