hawku/TabletDriver #1134: Driver stopped. Restarting! Check console!!!

Mojitoburrito01, 2021-06-20:
it doesnt detect my pen i cant do anything with the tablet this what comes on console:

New config created!
[2021-06-20 19:23:15.198] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:15.200] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:15.200] [ERROR] Possible fixes:
[2021-06-20 19:23:15.200] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:15.200] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:15.200] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:20.746] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:20.747] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:20.747] [ERROR] Possible fixes:
[2021-06-20 19:23:20.747] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:20.747] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:20.747] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:26.291] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:26.292] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:26.292] [ERROR] Possible fixes:
[2021-06-20 19:23:26.292] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:26.292] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:26.292] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:31.841] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:31.843] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:31.843] [ERROR] Possible fixes:
[2021-06-20 19:23:31.843] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:31.843] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:31.843] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:37.402] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:37.404] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:37.404] [ERROR] Possible fixes:
[2021-06-20 19:23:37.404] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:37.404] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:37.404] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:42.953] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:42.954] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:42.954] [ERROR] Possible fixes:
[2021-06-20 19:23:42.954] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:42.954] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:42.954] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:48.497] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:48.499] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:48.499] [ERROR] Possible fixes:
[2021-06-20 19:23:48.499] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:48.499] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:48.499] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:54.058] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:54.059] [ERROR] Can't open VMulti device!

[2021-06-20 19:23:54.059] [ERROR] Possible fixes:
[2021-06-20 19:23:54.059] [ERROR] 1) Install VMulti driver
[2021-06-20 19:23:54.059] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:23:54.059] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:23:59.609] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:23:59.611] [ERROR] Can't open VMulti device!

[... 99 lines cut ...]
[2021-06-20 19:25:06.279] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:25:11.840] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:25:11.842] [ERROR] Can't open VMulti device!

[2021-06-20 19:25:11.842] [ERROR] Possible fixes:
[2021-06-20 19:25:11.842] [ERROR] 1) Install VMulti driver
[2021-06-20 19:25:11.842] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:25:11.842] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2021-06-20 19:25:17.400] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2021-06-20 19:25:17.402] [ERROR] Can't open VMulti device!

[2021-06-20 19:25:17.402] [ERROR] Possible fixes:
[2021-06-20 19:25:17.402] [ERROR] 1) Install VMulti driver
[2021-06-20 19:25:17.402] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2021-06-20 19:25:17.402] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!

## Hardware/software information 
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: CTL-480

Kiried-Menacing, 2021-06-21:
Bro, it says to run the .bat file called "install_vmulti_driver.bat", hopefully this helps.
Also it keeps saying to install VMULTI DRIVER please learn to read, and if you already had other drivers, uninstall them and kill their processes in task manager or it wont work :)

SnowLire, 2021-06-22:
#1087