hawku/TabletDriver #1139: can't open vmulti device

w-i-s-n, 2021-06-24:
## Hardware/software information
Windows Version: windows 10

Tablet Driver version: 0.2.3

Tablet model: veikk s640
[image]

SnowLire, 2021-06-24:
this seems like a non-issue as the console has printed out your solution for you and all you need to do is read a bit, however #1087

w-i-s-n, 2021-06-24:
obviously i have done what was written and even more. in fact hawku kind of permanently caused problems i don't know why. now whenever an administrator window is on top of the desktop, the tablet just freezes until another window is on focus

w-i-s-n, 2021-06-24:
and yes i've switched back to the veikk drivers, however the difference is that i can no longer move the cursor on an administrator window, i tried opentablet and still the same, somehow something from hawku caused the tablet to lose admin privilages

w-i-s-n, 2021-06-24:
ive even also tried to remove and to reinsert device to no avail

w-i-s-n, 2021-06-24:
so a little help would be much appreciated rather than just state the obvious