hawku/TabletDriver #1143: the drivers say 1 monitor is 2

RandomWeeb115, 2021-06-27:
when i start my drivers, my scrren map area is set to 2nd screen, but when i use my pen it is on half of my 1st screen. i have tried a few things already but nothing has worked so far. please help this is fucking annoying

## Hardware/software information
Windows Version: windows 10

Tablet Driver version: v0.2.3

Tablet model: Wacom CTL-4100

TheRedEyedSpecter, 2021-06-29:
I think I have the same issue? I have 2 monitors. I only want to use one of them. Everything was fine with the mapping, but in recent times, now just about half of my screen is unreachable by my tablet. I cannot move the cursor over to that half of the screen using the tablet.

SnowLire, 2021-06-29:
#1087