hawku/TabletDriver #1151: can only use half of my screen now for some reason

TheBookOfSnaggy, 2021-07-05:
Alright so I have my usual drivers settings that I used for a year now and always worked fine until now. However, now my cursor whenever I use my tablet cannot cross the right hand half of my screen and it is not my tablet that is broken I tested some other drivers but I really wanna keep using this driver I hope you guys can help me fix this!

[image]

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G640

SnowLire, 2021-07-05:
known vmulti issue on Windows 10 21H1. Microsoft won't fix it so go use OpenTabletDriver

#1087

Ghost4815, 2021-07-13:
I had the same problem and there is actually a fix for this. Go into the tab "Misc" and under "Desktop Size" deselect "Auto" and enter the Width and Height of the monitor you want to use.

SnowLire, 2021-07-14:
why do you people still want to use an outdated, unsupported driver?

BeigeAnimal, 2021-07-22:
> why do you people still want to use an outdated, unsupported driver?

I need it because I use a 420 and the stock drivers are shit. You can't force the aspect ratio so you just have to eyeball it.
I'd get a better tablet but I don't have money, and I only use it for osu.

SnowLire, 2021-07-22:
Fair, although all non-wacom OEM drivers are also broken because they use vmulti just like HawkuTD