hawku/TabletDriver #1168: xp-pen deco mini4

IIIBopoHIII, 2021-08-23:
## Hardware/software information
Windows Version: 10(64)

Tablet Driver version: ?

Tablet model: xp-pen deco mini4

[2021-08-23 10:59:18.841] >> DetectMask 0x40
[2021-08-23 10:59:18.841] >> MaxX 21600
[2021-08-23 10:59:18.841] >> MaxY 13500
[2021-08-23 10:59:18.841] >> MaxPressure 1023
[2021-08-23 10:59:18.841] >> Width 216.000
[2021-08-23 10:59:18.841] >> Height 135.000
[2021-08-23 10:59:18.841] >> InitFeature 0x02 0x02
[2021-08-23 10:59:18.841] >> # Wacom CTH-460
[2021-08-23 10:59:18.841] >> HIDTablet 0x056A 0x00D1 0xFF0D 0x0001
[2021-08-23 10:59:18.842] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D1 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.842] >> HIDTablet 0x056A 0x00D1 0x000D 0x0001
[2021-08-23 10:59:18.843] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D1 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.843] >> Name "Wacom CTH-460"
[2021-08-23 10:59:18.843] >> ReportLength 9
[2021-08-23 10:59:18.843] >> DetectMask 0x40
[2021-08-23 10:59:18.843] >> MaxX 14720
[2021-08-23 10:59:18.843] >> MaxY 9200
[2021-08-23 10:59:18.843] >> MaxPressure 1023
[2021-08-23 10:59:18.843] >> Width 147.200
[2021-08-23 10:59:18.843] >> Height 92.000
[2021-08-23 10:59:18.843] >> InitFeature 0x02 0x02
[2021-08-23 10:59:18.843] >> # Wacom CTH-460(A)
[2021-08-23 10:59:18.843] >> HIDTablet 0x056A 0x00D6 0xFF0D 0x0001
[2021-08-23 10:59:18.844] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D6 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.844] >> HIDTablet 0x056A 0x00D6 0x000D 0x0001
[2021-08-23 10:59:18.845] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D6 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.845] >> Name "Wacom CTH-460(A)"
[2021-08-23 10:59:18.845] >> ReportLength 9
[2021-08-23 10:59:18.845] >> DetectMask 0x40
[2021-08-23 10:59:18.845] >> MaxX 14720
[2021-08-23 10:59:18.845] >> MaxY 9200
[2021-08-23 10:59:18.845] >> MaxPressure 1023
[2021-08-23 10:59:18.845] >> Width 147.200
[2021-08-23 10:59:18.845] >> Height 92.000
[2021-08-23 10:59:18.845] >> InitFeature 0x02 0x02
[2021-08-23 10:59:18.845] >> # Wacom CTH-461
[2021-08-23 10:59:18.845] >> HIDTablet 0x056A 0x00D2 0xFF0D 0x0001
[2021-08-23 10:59:18.846] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D2 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.846] >> HIDTablet 0x056A 0x00D2 0x000D 0x0001
[2021-08-23 10:59:18.847] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D2 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.847] >> Name "Wacom CTH-461"
[2021-08-23 10:59:18.847] >> ReportLength 9
[2021-08-23 10:59:18.847] >> DetectMask 0x40
[2021-08-23 10:59:18.847] >> MaxX 14720
[2021-08-23 10:59:18.847] >> MaxY 9200
[2021-08-23 10:59:18.847] >> MaxPressure 1023
[2021-08-23 10:59:18.847] >> Width 147.200
[2021-08-23 10:59:18.847] >> Height 92.000
[2021-08-23 10:59:18.847] >> InitFeature 0x02 0x02
[2021-08-23 10:59:18.847] >> # Wacom CTH-461(A)
[2021-08-23 10:59:18.847] >> HIDTablet 0x056A 0x00D7 0xFF0D 0x0001
[2021-08-23 10:59:18.848] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D7 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.848] >> HIDTablet 0x056A 0x00D7 0x000D 0x0001
[2021-08-23 10:59:18.849] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00D7 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-08-23 10:59:18.849] >> Name "Wacom CTH-461(A)"
[2021-08-23 10:59:18.849] >> ReportLength 9
[2021-08-23 10:59:18.849] >> DetectMask 0x40
[2021-08-23 10:59:18.849] >> MaxX 14720
[2021-08-23 10:59:18.849] >> MaxY 9200
[2021-08-23 10:59:18.849] >> MaxPressure 1023
[2021-08-23 10:59:18.849] >> Width 147.200
[2021-08-23 10:59:18.849] >> Height 92.000
[2021-08-23 10:59:18.849] >> InitFeature 0x02 0x02
[... 214 lines cut ...]
[2021-08-23 10:59:18.877] [STATUS] COMMANDS CheckDeviceString CheckString CheckTablet ClearAuxButtonMap ClearAuxCustomData ClearButtonMap 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS ClearCustomData ClickPressure CloseTablet CustomDataInstruction AuxCustomData CustomData 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS Debug DesktopSize Desktop DetectMask Echo Exit Quit FilterTester Tester FilterTimerInterval Interval TimerInterval 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS GetCommands GetDeviceStrings GetString GetStrings GetFeature GravityFilter Gravity 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS Height Help HIDList HIDTablet IgnoreMask Include Information Info InitFeature InitReport 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS InitStrings InitString KeepTipDown ListCommands List ListKeys Log LogDirect LogTabletArea 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS MaxPressure MaxX MaxY Measure NoiseReduction Noise Output OutputMode Mode OutputReport Report 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS PressureDeadzone PressureSensitivity RelativeResetDistance ResetDistance RelativeResetTime ResetTime 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS RelativeSensitivity Sensitivity ReportId ReportLength Rotate ScreenArea ScrollAcceleration ScrollAcc 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS ScrollSensitivity ScrollStopCursor SetFeature Feature Skew SmoothingFilter Smoothing 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS Start Status TabletArea Area TabletDataFormat Format TabletFormat TabletMove Move 
[2021-08-23 10:59:18.877] [STATUS] COMMANDS TabletName Name TabletValid USBTablet USBWrite Wait Width 
[2021-08-23 10:59:18.877] 
[2021-08-23 10:59:18.877]   Driver version: 0.2.3
[2021-08-23 10:59:18.877]   Windows version: Microsoft Windows NT 6.2.9200.0
[2021-08-23 10:59:18.877]   Windows product: Windows 10 Enterprise
[2021-08-23 10:59:18.877]   Windows release: 2009
[2021-08-23 10:59:18.877] 
[2021-08-23 10:59:18.878] >> CheckTablet
[2021-08-23 10:59:18.878] [ERROR] Tablet not found!
[2021-08-23 10:59:18.878] [ERROR] Check the list of supported tablets from the GitHub page.
[2021-08-23 10:59:18.878] [ERROR] http://github.com/hawku/TabletDriver
Driver stopped. Restarting!

Please help fix problem

IIIBopoHIII, 2021-08-23:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/7030161/startuplog.txt)

SnowLire, 2021-08-23:
#1087

jamesbt365, 2021-08-23:
Pop over to the OpenTabletDriver discord server and we can get you support entirely