hawku/TabletDriver #1177: Pentablet driver restarting on loop no tablet found.

AWildMat, 2021-09-18:
## Hardware/software information
Windows Version: 10.0.19043 Build 19043

Tablet Driver version: 
0.2.3
Tablet model: 
CTL 472
-----
I am new to tablet so excuse me for any mistakes or lack of knowledge.
I have bought the ctl 472 a while back and when installing the drivers everything worked perfectly even when I had my old xp pen drivers installed. 

Fast forward to one motherboard change and PC format later I installed the drivers again and everything worked fine however I was waiting for my tablet cover so I did not use the tablet until the cover arrived (1 month) however when opening the driver up I had issues with the tablet cursor stopping around 2/3 of the way on screen even though my screen map was set to the right resolution and the right monitor. I reinstalled the driver using the remove_vmulti_driver.bat and restarted my pc. When installing the driver again I had issues setting my area and I cannot set the tablet to a single monitor anymore it's stuck being able to move around on all 3 and even though it is in absolute (osu!) mode it acts as it's in relative mouse mode. 

Furthermore when looking through forums I found various fixes through using device manager, updating the wacom tablet etc etc. however my device manager does not even show wacom tablet as an option (Picture for reference) and my console for tablet driverGUI is filled with errors (Another picture for reference). 

Sorry for the long text but i tried to be thorough with my issue. 

[image]

[image]

SnowLire, 2021-09-18:
Don't use HawkuTD anymore. Switch to OpenTabletDriver #1087

AWildMat, 2021-09-18:
> Don't use HawkuTD anymore. Switch to OpenTabletDriver #1087

May I ask how come, Is there a specific reason?

jamesbt365, 2021-09-18:
Because hawku hasnt been updated in over 3 years and is exteremely broken whereas otd has better features, support and gets updated