hawku/TabletDriver #1183: Tablet doesnt output anything

skittduge, 2021-10-03:
So my tablet doesnt output anything it just doesnt detect any movment, I do not have installed any other drivers, and the latest version of this driver. and the tablet works normaly with the wacom driver.

*** Console log: 

[2021-10-03 20:15:24.833] >> HIDTablet 0x056A 0x0314 0xFF0D 0x0001
[2021-10-03 20:15:24.835] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0314 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.835] >> HIDTablet 0x056A 0x0314 0x000D 0x0001
[2021-10-03 20:15:24.837] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0314 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.837] >> Name "Wacom PTH-451"
[2021-10-03 20:15:24.837] >> ReportLength 10
[2021-10-03 20:15:24.837] >> DetectMask 0x40
[2021-10-03 20:15:24.837] >> MaxX 31496
[2021-10-03 20:15:24.837] >> MaxY 19685
[2021-10-03 20:15:24.837] >> MaxPressure 2047
[2021-10-03 20:15:24.837] >> Width 157.480
[2021-10-03 20:15:24.837] >> Height 98.425
[2021-10-03 20:15:24.837] >> InitFeature 0x02 0x02
[2021-10-03 20:15:24.837] >> TabletFormat WacomIntuosV2
[2021-10-03 20:15:24.837] >> # Wacom PTH-650
[2021-10-03 20:15:24.837] >> HIDTablet 0x056A 0x0027 0xFF0D 0x0001
[2021-10-03 20:15:24.838] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0027 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.838] >> HIDTablet 0x056A 0x0027 0x000D 0x0001
[2021-10-03 20:15:24.840] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0027 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.840] >> Name "Wacom PTH-650"
[2021-10-03 20:15:24.840] >> ReportLength 10
[2021-10-03 20:15:24.840] >> DetectMask 0xE0
[2021-10-03 20:15:24.840] >> MaxX 44704
[2021-10-03 20:15:24.840] >> MaxY 27940
[2021-10-03 20:15:24.840] >> MaxPressure 2047
[2021-10-03 20:15:24.840] >> Width 223.520
[2021-10-03 20:15:24.840] >> Height 139.700
[2021-10-03 20:15:24.840] >> InitFeature 0x02 0x02
[2021-10-03 20:15:24.840] >> TabletFormat WacomIntuosV2
[2021-10-03 20:15:24.840] >> # Wacom PTH-651
[2021-10-03 20:15:24.840] >> HIDTablet 0x056A 0x0315 0xFF0D 0x0001
[2021-10-03 20:15:24.841] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0315 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.842] >> HIDTablet 0x056A 0x0315 0x000D 0x0001
[2021-10-03 20:15:24.843] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0315 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.843] >> Name "Wacom PTH-651"
[2021-10-03 20:15:24.843] >> ReportLength 10
[2021-10-03 20:15:24.843] >> DetectMask 0x40
[2021-10-03 20:15:24.843] >> MaxX 44704
[2021-10-03 20:15:24.843] >> MaxY 27940
[2021-10-03 20:15:24.843] >> MaxPressure 2047
[2021-10-03 20:15:24.843] >> Width 223.520
[2021-10-03 20:15:24.843] >> Height 139.700
[2021-10-03 20:15:24.843] >> InitFeature 0x02 0x02
[2021-10-03 20:15:24.843] >> TabletFormat WacomIntuosV2
[2021-10-03 20:15:24.843] >> # Wacom PTH-850
[2021-10-03 20:15:24.843] >> HIDTablet 0x056A 0x0028 0xFF0D 0x0001
[2021-10-03 20:15:24.845] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0028 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.845] >> HIDTablet 0x056A 0x0028 0x000D 0x0001
[2021-10-03 20:15:24.846] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0028 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.846] >> Name "Wacom PTH-850"
[2021-10-03 20:15:24.846] >> ReportLength 10
[2021-10-03 20:15:24.846] >> DetectMask 0xE0
[2021-10-03 20:15:24.846] >> MaxX 65024
[2021-10-03 20:15:24.846] >> MaxY 40640
[2021-10-03 20:15:24.846] >> MaxPressure 2047
[2021-10-03 20:15:24.846] >> Width 325.120
[2021-10-03 20:15:24.846] >> Height 203.200
[2021-10-03 20:15:24.846] >> InitFeature 0x02 0x02
[2021-10-03 20:15:24.846] >> TabletFormat WacomIntuosV2
[2021-10-03 20:15:24.846] >> # Wacom PTH-851
[2021-10-03 20:15:24.846] >> HIDTablet 0x056A 0x0317 0xFF0D 0x0001
[2021-10-03 20:15:24.848] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0317 UsagePage=0xFF0D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.848] >> HIDTablet 0x056A 0x0317 0x000D 0x0001
[2021-10-03 20:15:24.850] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x0317 UsagePage=0x000D Usage=0x0001 Exclusive=False
[2021-10-03 20:15:24.850] >> Name "Wacom PTH-851"
[2021-10-03 20:15:24.850] >> ReportLength 10
[... 216 lines cut ...]
[2021-10-03 20:15:42.973] >> FilterTimerInterval 10
[2021-10-03 20:15:42.973] Filter Timer Interval = 10 (100.00 Hz)
[2021-10-03 20:15:42.973] >> Smoothing 0
[2021-10-03 20:15:42.973] Smoothing = off
[2021-10-03 20:15:42.973] >> Noise 0
[2021-10-03 20:15:42.973] Noise Reduction = off
[2021-10-03 20:15:42.973] >> AntiSmoothing 0
[2021-10-03 20:15:42.973] Anti-smoothing = off
[2021-10-03 20:15:42.973] >> Debug false
[2021-10-03 20:15:42.973] Debug logging = False
[2021-10-03 20:21:11.299] >> Debug true
[2021-10-03 20:21:11.300] Debug logging = True
[2021-10-03 20:23:01.275] >> Debug false
[2021-10-03 20:23:01.275] Debug logging = False
[2021-10-03 20:23:05.602] >> Debug true
[2021-10-03 20:23:05.602] Debug logging = True
[2021-10-03 20:23:07.030] >> Debug false
[2021-10-03 20:23:07.030] Debug logging = False

## Hardware/software information
Windows Version: Windows 10 Home, version 2009

Tablet Driver version: 0.2.3

Tablet model: MTE-450

SnowLire, 2021-10-03:
Don't use hawkuTD. Your problem won't be fixed ever. Use OpenTabletDriver instead #1087

skittduge, 2021-10-03:
I wanna use this one since on OpenTabletDriver I can't use pressure sensors
and the don't want to fix it either, just no response whatsoever

SnowLire, 2021-10-04:
You can get pressure sensitivity on OTD. You need to download the WindowsInk plugin and its requirement. Everything you need is in the plugin's wiki page.

skittduge, 2021-10-04:
Look if your not a developer there, you cant help me cuz I already talked to them. And they tried, but then just vanished. So I already tried everything. It must be a software problem.

SnowLire, 2021-10-04:
Can you give your discord username so I can search back in the message logs to see what your exact problem with it was? Thanks.

skittduge, 2021-10-04:
Why would you need my discord username? you can search it by my GitHub name plus  there is no reason for you to help me.