hawku/TabletDriver #1200: Driver Stopped restarting check console

wgfesgr, 2021-12-08:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: newest

Tablet model: Veikk S640 
[image]

jamesbt365, 2021-12-11:
#1087

GasparAraneta, 2021-12-24:
i have the same issue with the same tablet model

VB304, 2022-03-30:
Reinstalled windows 10 and did the same configuration and still works for my Veikk S640.

I don't know which one that did it but I did:

Fully enabled the .Net framework 3.5 and 4.8 in windows feature.
Installed veikk drivers but not in Startup

[image]