hawku/TabletDriver #1216: Помогите решить проблему с планшетом parblo a640 v2

BlackHoliday, 2022-03-29:
Драйвер hawku в отличии от фирменного драйвера pablo не читает мой планшет parblo a640 v2, что мне нужно добавить в конфиг чтобы драйвер прочитал мой планшет?
Помогите как можно скорее, я очень прошу вас, устал бороться с задержкой как минимум

jamesbt365, 2022-03-30:
#1087
https://github.com/hawku/TabletDriver/blob/master/README.md

BlackHoliday, 2022-03-30:
I have 32 bit window system, and i cant use opentablet

jamesbt365, 2022-03-30:
This driver has no support anyway, it hasn't been updated in years, you will get no support from the people who made it and therefore you'll have to use opentabletdriver or you'll be stuck with the OEM Driver for the tablet. I can send you build instructions on how to build a 32 bit copy of opentabletdriver, however it is not an officially supported architechure, i reccomend you reinstall your operating system with 64 bit. However if you truly want to go ahead and build your own copy of a driver that has 32 bit support then follow this.

- Download and install [this](https://dotnet.microsoft.com/en-us/download/dotnet/thank-you/sdk-6.0.201-windows-x86-installer)
- download git then and git clone https://github.com/opentabletdriver/opentabletdriver, alternatively download master from [here](https://github.com/OpenTabletDriver/OpenTabletDriver/archive/refs/heads/master.zip). (if you download the second one you'll need to extract it first)
- Open up a CMD window if you haven't already, `cd Directory/of/opentabletdriver/extracted` (you can copy the directory in file explorer)
- type these 2 commands after you are in the correct directory.
- `dotnet build OpenTabletDriver.Daemon/OpenTabletDriver.Daemon.csproj --configuration Release --runtime win-x86 --framework net6 -o build/win-x86`
- `dotnet build OpenTabletDriver.UX.Wpf/OpenTabletDriver.UX.Wpf.csproj --configuration Release --runtime win-x86 --framework net6-windows -o build/win-x86`
- once you have done that you'll get a build folder which will have the exes for opentabletdriver, both `OpenTabletDriver.Daemon.exe` and `OpenTabletDriver.UX.Wpf.exe`

Do keep in mind i cannot build windows binaries because you can't build wpf on linux for windows, so you will have to do that yourself, and you will not get support for opentabletdriver 32 bit binaries so don't make a bug report on the repo with this in mind, however your tablet should detect after this, if it doesn't then change the config by checking a diag for what your tablet identifier is.

BlackHoliday, 2022-03-30:
Thaka so much for the support