hawku/TabletDriver #1221: XP-Pen G640S having issues with the driver

Hokurikudo, 2022-05-27:
I've downloaded the drivers as shown in the installation guide, but even though I've installed it correctly, it shows me this error on the bottom left, "Driver stopped. Restarting! Check console !!!" and when the console is checked it shows,

"Tablet init failed!
Possible fixes:

    1.Uninstall other tablet drivers.
    2.Stop other tablet driver-services."

I have uninstalled my previous driver which was the XP-Pen driver and I'm unsure why the hawku drivers won't work. Thank you so much for taking your time to read this! :)

Windows Version: Windows 7

Tablet Driver version: V0.2.3

Tablet model: XP-Pen G640S

jamesbt365, 2022-05-27:
read the readme

Hokurikudo, 2022-05-28:
> read the readme

no driver for windows 32-bit?

jamesbt365, 2022-05-30:
Wrong: https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/2393682240
Also this driver doesn't have 32 bit binaries anyway.