hawku/TabletDriver #1234: Help man

Nuruji, 2023-07-19:
How do I uninstall both the hawku drivers and open tablet drivers?
They dont show up in application remover????

I just want to play osu and my tablet not working sadge 

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Nuruji, 2023-07-20:
I have a gaomon s620 and it was working on hawku driver before. I eft my tablet unplugged for a while and took a break from osu and now it dont work. My boomer brain wants to say its because I updated my laptop but im unsure. If anyone could help with that too that'd be cool too.

jamesbt365, 2023-09-01:
Don't use hawku, use OpenTabletDriver instead, hawku is broken. #1087.