hawku/TabletDriver #1243: I dont know how to install open tablet file

Adrian32235, 2024-11-23:
Hey guys, yea so like i installed opentablet driver from github however i dont know how to extract it or how to make the files for it, there is no configurations for it nor the things that should been. 
can i get some help with it i want it to work and i dont know how to make it work, im a slow person so like if you know how to make it please add my discord and help me with sharescreen. God bless

Discord Username: adrian26l

Tablet Model: XP Pen Star G640

Adrian32235, 2024-11-23:
[image]

0arin, 2025-02-18:
hi I sent you a friend request you seem to be confused. I have a wacom but I dont see how it would be different